dppdppd/The-Boardgame-Insert-Toolkit #2: Margin in another component.

MikiGrit, 2020-02-28:
Hi,

I'm trying to do a boardgame insert with your toolkit (which is great btw, thanks for it! :) ).

And I'm facing difficulty how to do particular thing. I want to have adjusted padding height in top right corner of insert, so there is a space for some square boards on top of other components. But instead of cropping the padding between leftside comparments, it inserts rectangle into them. Am I doing it wrong or is this currently not possible?

Thanks!

I have this configuration:
```

include <boardgame_insert_toolkit_lib.2.scad>;

// determines whether lids are output.
g_b_print_lid = f;
// determines whether boxes are output.
g_b_print_box = true;

// Focus on one box
g_isolated_print_box = "insert";

// Used to visualize how all of the boxes fit together.
g_b_visualization = false;

// this is the thickness of partitions between compartments
// Default = 1
g_partition_thickness = 2.0;

// this is the width of partitions that are for 
// inserting fingers to grab the bits.
// default = 13
g_finger_partition_thickness = 13; 

// this is the outer wall thickness.
//Default = 1.5mm
g_wall_thickness = 2.0;

// tolerance for fittings. This is the gap between fitting pieces,
// such as lids and boxes. Increase to loosen the fit and decrease to
// tighten it.
// Default = 0.1mm
g_tolerance = 0.2;

// Lid

// this is the lid thickness.
// Default = g_wall_thickness
//g_lid_thickness = 1.5;

// the depth of the lid
g_lid_lip_height = 5.0; // default = 5.0

// creates the indentation on the bottom of the box 
//that allows the lid to be put under when in play.
g_b_fit_lid_underneath = 1;

// Makes solid simple lids instead of the honeycomb ones.
// Might be faster to print. Definitely faster to render.
g_b_simple_lids = false;

function _compartment_text(labels) = ["label",
    [
        ["text", labels],
        [ "placement", "center"],
        [ "rotation", 10],
        [ "font_size", "auto"],
        [ "offset", [0, 0]],
        [ "font", "Iosevka:style=bold italic"],
    ]
];

 
data =
[
    [   "insert",
        [
            [ "box_dimensions", [193, 193, 44] ],
            [ "label",
                [
                    [ "text", "my insert"],
                    [ "width", "auto"],
                    [ "rotation", 45],
                    [ "offset", [-4,-2]],
                ]
            ],

            [ "components",
                [
                    [   "cards",
                        [
                            ["num_compartments", [1,2]],
                            ["compartment_size", [94, 60, 20]],
                            ["padding",                         [22, 2]],
                            ["margin",                          [f,f,t,f]],
                            ["padding_height_adjust",           [-7, -7] ],
                            ["cutout_sides",                    [t,f,f,f]],
                            ["position",                        ["max","max"]],
                        ]
                    ],
                    [   "slides",
                        [
                            ["num_compartments", [1,2]],
                            ["compartment_size", [93, 93, 40]],
                            ["padding", [2, 2]],
                            ["position", [0, 0]],
                            ["cutout_sides", [t,f,f,f]],
                            _compartment_text([["cubes"], ["dice"]])
                        ]
                    ],
                ]
            ]
        ]
    ],
];

MakeAll();
```

IdoMagal, 2020-02-29:
I've changed the main loop so that now components are constructed as negative shapes and then finally subtracted from the box. What this means is that you can intersect components to carve out secondary shapes, such as the square boards you're trying to fit on top. Let me know if this does what you're intending. Try the following:

```
data =
[
    [   "insert",
        [
            [ "box_dimensions", [193, 193, 44] ],
            [ "label",
                [
                    [ "text", "my insert"],
                    [ "width", "auto"],
                    [ "rotation", 45],
                    [ "offset", [-4,-2]],
                ]
            ],

            [ "components",
                [
                    [   "cards",
                        [
                            ["num_compartments", [1,2]],
                            ["compartment_size", [94, 60, 20]],
                       //     ["padding",                         [22, 2]],
                            ["margin",                          [f,f,t,f]],
                            ["padding_height_adjust",           [-7, -7] ],
                            ["cutout_sides",                    [t,f,f,f]],
                            ["position",                        ["max","max"]],
                        ]
                    ],
                    [   "slides",
                        [
                            ["num_compartments", [1,2]],
                            ["compartment_size", [93, 93, 40]],
                            ["padding", [2, 2]],
                            ["position", [0, 0]],
                            ["cutout_sides", [t,f,f,f]],
                            _compartment_text([["cubes"], ["dice"]])
                        ]
                    ],

                    [ "square boards",
                    [
                        ["position",                        ["max","max"]],
                        ["compartment_size", [116, 140, 7]],

                    ]]
                ]
            ]
        ]
    ],
];
```

MikiGrit, 2020-02-29:
Awesome! It also feels more intuitive now. Thanks :).

closed 2020-02-29 by IdoMagal