dppdppd/The-Boardgame-Insert-Toolkit #6: v2.12+ Lid Detents

mdoom23, 2020-06-12:
With v2.12, the tolerance between boxes and lids was increased and the detents feature implemented.

I think this is a great idea because I often struggled previously with too tight of fight (and I never messed with g_tolerance from default).

Today I printed my first box using this new functionality.  Possibly because of having side cut outs in the box thus removing all detents from the sides, however with default settings its an excessively loose box and I cant keep the lid on.

And this is even after increasing my g_wall_thickness from 1.5 up to 2.  

Possibly the radius on them should be more than 1/3 of m_wall_thickness?

Possibly on any sides that have "cutouts" position potentially needs to be modified based on the number of compartments there are on that axis?  Per attached images you can see detents are quite small, as well as on the longer sides of my example box, the detents on the lid have nothing to hold onto as they are directly in the middle of the cutouts.
[image]
[image]
[image]

IdoMagal, 2020-06-12:
Makes sense. Can you do me a favor and print out the simple box? I’d like to first confirm that the detents, as designed, work for you. If they don’t then I think the solution would be more involved than if it’s just a matter of making sure there enough of them.

mdoom23, 2020-06-12:
Sure thing! I'll try that right away

mdoom23, 2020-06-13:
Well, my 30x30 simple box... i can't get the lid off at all! 🤣 

I do notice that with the standard g_wall_thickness of 1.5, that the holes for the detents are always entirely through the wall of the box.  However with my g_wall_thickness of 2 it wasn't.

I suppose that makes sense since radius is 1/3 of thickness, but the 1/2 of thickness thats taken out for space for the lid, has a different amount of delta between the two.

For the simple box, I switched to very basic Hatchbox PLA.  nothing fancy.  printed at 0.3 mm.  Should be fairly consistent quality.
[image]

mdoom23, 2020-06-13:
I'll gladly help how i can this weekend on trying different things.  I've definitely dug into your code plenty! 

If its a matter of how many detents, personally i'd take a strategy of it being based on a variable.  to essentially represent max distance between dents..sort of.    say, 40 mm for default (lets call it x for arguments sake).  in my mind i'd look at each side of the box dims, and do something like  box_dimension // x + 1.  So, integer division.  Meaning, with 40, any side with 1-39 mm would result in 1 detent.  anything with 40 - 79, result in two. etc.

Then the complicated part is dealing with the array of components and if any of them have cutouts, and if so where they are located.  And if you offset/adjust detents or not.  

Purely brainstorming here.   Only because I think my issue with the 30x30 box i printed was that 2 on each side was simply too many and thats why i can't get it open! 🤣

IdoMagal, 2020-06-13:
OK. 2 detents on a 30x30 with no notches is unopenable for you too. :)

I would like to solve the extreme case of the box with only two opposing sides holding the lid because I know I've made boxes like that (with 2 whole sides cut out). 
So I think the task at hand is to see what combination of detents and looseness works there.

mdoom23, 2020-06-13:
Submitted a pull request.  Doesn't explicitly solve the issue, however it does allow the end-user to have full control and flexibility.  Feel free to modify as you see fit!

IdoMagal, 2020-06-13:
I haven't had a chance to look at your PR, but I think v2.16 may be comparable? The salient feature here is that I added more approximate detents and I added them to the corners. The advantage of the corners is that the flexion is going to be more similar regardless of how big the box is. It seems to work well with my test cases. Can you test with yours and report back?

mdoom23, 2020-06-15:
I skimmed your code changes. Haven't tried generating anything yet, will today.  Interesting take with the focus being on the corners.  Definitely makes sense.

Yeah feel free to check out what i proposed.  Different approach, sort of.   More about just having dynamic counts.  I didn't do it based on corners though and made each side independent.

I'm excited to check out some tests though with your version with the changes to the ratios and size of the detents to see how they impact things across the board.

mdoom23, 2020-06-20:
Sorry for the delay, busy week!  Anyways, I printed a new box, slightly smaller than the last one, and now with v2.16.  It definitely isn't too loose this time.  Fairly tight fit.  Personally so far I'm not sure its any different / better than original prior to detents.

The detents do make for a very strong bond in the corners when you put the lid underneath the box where there aren't detent openings.  Almost too strong.  Please see the pics.  Don't mind my quick and dirty print settings on this one!

[image]

[image]

IdoMagal, 2020-06-20:
Ok. Good? The attempt was to address the unreliability of the old friction method. So hopefully this one will be more consistent. Time will tell. If you experience poor closure on a particular design, let me know. We can also revert or make it optional.

mdoom23, 2020-06-22:
So, I printed a few more boxes of different shapes and tested those with lids now too.  All in all, I think the current detents works well. I haven't tried playing with the g_tolerance though to see how that impacts them.

Anyways though, I do think we need detent holes in the bottom of the lid too if LID_FIT_UNDER_B is set to true.  Especially on smaller boxes, the lid will get permanently stuck on the bottom without them currently.  If we add that, I think we can call it a perfect v2.17 and then I'll publish the insert that I've been working on through all of this. 😃

IdoMagal, 2020-06-22:
Done.

mdoom23, 2020-06-23:
Thank you!!!!
I think this issue can be closed. 😃 

I published my insert using all v2.17 STL files.  Successfully as a remix on thingiverse too.

[https://www.thingiverse.com/thing:4493482](https://www.thingiverse.com/thing:4493482)

closed 2020-07-08 by IdoMagal