dppdppd/The-Boardgame-Insert-Toolkit #10: Feature request: Option to render labels seperately for multiple extruders

takras, 2020-08-24:
First off, thanks for this wonderful tool. I've made a few inserts based off your previous version (sword & sorcery one for instance), and will continue to use it.

I've tried to alter the source code and contribute that way, but I'm not a maths or geometry person, so had to give up. I do have a feature request.

I use a Prusa MMU and have imported (using BIT v1) the rendered stl's to Tinkercad and dome some manipulation there to separate the label from the box and lid, and placed in it's own stl object. This way I can import it into the slicer and apply different filaments for them, and having the text as another color.

Would it be possible to have an option for this, to render the text as seperate objects, while maintaining the placeholder for the text on the lid/box?

Screenshot from my tinkercad trying to illustrate what I mean:
[image]

The text has been stripped from the lid and seperated to its own layer. For the boxes, I've inverted the hollow text into their own objects, so when printing, it will be filled with its own filament color.

IdoMagal, 2020-08-25:
Hey. Thanks for the feedback. That sounds like a fine addition. I'll investigate.

IdoMagal, 2020-08-25:
Checked in with version v3.30. Let me know if that works or if I missed something.

Cheers.

takras, 2020-08-25:
That was very quickly delivered, awesome.

It works fine for the lid, but the box now seems to be broken. The labels appear to stick to the bottom now instead of where it's supposed to be.

Here's the box with **g_print_mmu_layer = "default";**
[image]

[image]

`[ BOX_SIZE_XYZ,                     [70, 206, 32] ],
            [ ENABLED_B, true],
            [ BOX_COMPONENT, [
                [CMP_COMPARTMENT_SIZE_XYZ,      [67, 67, 31]],
                [ CMP_NUM_COMPARTMENTS_XY,      [1, 3] ],
                [CMP_SHAPE,                     FILLET],
                [CMP_FILLET_RADIUS,             5],
                [LABEL,               
                        [
                            [LBL_TEXT,        [   
                                                ["Test"],
                                                ["Test"],
                                                ["Test"]
                                            ]
                            ],
                            [ LBL_PLACEMENT,     CENTER],
                            [ ROTATION,         20],
                            [ LBL_SIZE,         7],
                            [ POSITION_XY,      [ 0,0]],
                            [ LBL_DEPTH, 1 ]
                        ],
                    ], 
            ]]`

IdoMagal, 2020-08-25:
That’s probably related to 2.28. I’ll have a look tomorrow.

IdoMagal, 2020-08-26:
fixed in v2.31

closed 2020-08-26 by IdoMagal