dppdppd/The-Boardgame-Insert-Toolkit #11: v2.30 - Component Labels

mdoom23, 2020-08-25:
Two fold,
I don't believe a new issue, but I just noticed with latest version.  I was using the LBL_PLACEMENT attribute of component labels. 

When using RIGHT_WALL or LEFT_WALL it currently is just moving the text over to the walls, but not doing an implied z-axis rotation of 90/-90 degrees to actually put text on side walls. See attached screenshot of the simple box with a label on RIGHT_WALL.

In addition, I know v2.08 functionality was added to support multiple labels on lids.  Is there an appetite for adding this functionality on components as well?

Ultimately I'm attempting to put labels on the outsides of boxes, and using components to do so.

[image]

mdoom23, 2020-08-25:
Extra info - Just to show what I've been doing with multiple components (and moving one outside the position of box) to get multiple 'box' labels.
[image]

IdoMagal, 2020-08-26:
The bug is fixed in 2.32.

I'll look into adding more label support.

IdoMagal, 2020-08-26:
multi-label compartment support added in 2.33

closed 2020-08-26 by IdoMagal

IdoMagal, 2020-08-26:
Added ability to label the box itself with v2.34

Cheers.

mdoom23, 2020-08-26:
Thanks for the quick work!