dppdppd/The-Boardgame-Insert-Toolkit #13: v2.35 - Box Labels - BACK depth inverted

mdoom23, 2020-08-26:
The Depth on box labels when put on the 'BACK' of back seems to extrude outward from the box. I think the direction just needs to be inverted.

You can see in screenshot the lack of depth on the one side.  When exported then it becomes just a flat wall.  Just a simple box test
[image]

IdoMagal, 2020-08-27:
The example box renders correctly for me.

Can you 

1. render the box "labels" from the examples
2. post your box here.

[image]

mdoom23, 2020-08-27:
So good news bad news...

I get same results as you.  Example works fine.  The only things unique in mine was wall thickness of 2 instead of 1.5, and depth of .8 instead of .2, however I tested all combinations and they rendered perfectly.

I narrowed it down to not being about the 'BACK' object as initially figured.  It seems to be that specific character in the font I am using.

However, when I just generate a base text object with that font of that character, it generates identical to any other character/font.  So i'm baffled where it could be failing.

text("C", font = "Freehand Spirit:style=Regular");
[image]

So, all in all, I think my problem is somehow due to the font I'm using.  Although still baffling how its just that character, but it generates as a normal object.  

I generated a box with every character from this font as a box label, and it truly is limited to that single character.  So, sounds like problem is on my end. :-)  But all in all, still confusing when it generates as text object.
[image]

IdoMagal, 2020-08-27:
That's unfortunate. You could try hacking the $fn to higher or lower values and see if you get different results. There might be something about that vector that it doesn't like to difference().

mdoom23, 2020-08-27:
Good call!  Lower $fn did the trick. Didn't even think about that.  Thanks again for your help.  I have no shortage of more ideas for enhancements.  Hope you don't mind all these submissions.  I love how quick you're improving the library.

closed 2020-08-27 by mdoom23

IdoMagal, 2020-08-27:
Glad that worked. Feel free to request all of the features. The worst that could happen is nothing.