dppdppd/The-Boardgame-Insert-Toolkit #14: Slight overhang in cutouts when creating Stacked Boxes

Alatarvinyamar, 2020-09-08:
Hope I'm not missing anything obvious here:

[   "TERRAIN",
        [
            [ BOX_SIZE_XYZ,                                     [95, 70, 5.0] ],
            [ BOX_LID, 
                [ LID_INSET_B, f],
            ],
            [ BOX_NO_LID_B, t ],         
            [ BOX_STACKABLE_B, t],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 91, 66, 3] ],
                    [CMP_NUM_COMPARTMENTS_XY,               [1,1 ]], 
                    [CMP_CUTOUT_SIDES_4B,                   [t,t,f,f]],
            
                    ["label",
                        [
                           ["text",        "TERRAIN"],
                           [ LBL_FONT,         "Space Marine"],
                           ["size",        5],                                
                         ]
                     ]
                ]
            ],                            
        ]
    ],

IdoMagal, 2020-09-08:
Nope. Fixed in 2.38.

closed 2020-09-08 by IdoMagal

Alatarvinyamar, 2020-09-08:
ERROR: Parser error in file "E:/My Documents/3D Printing/Space Hulk Death Angel/boardgame_insert_toolkit_lib.2.scad", line 6: syntax error 
Execution aborted

IdoMagal, 2020-09-08:
That doesn't reproduce for me. Can you post a simple scad file that reproduces the issue?

Alatarvinyamar, 2020-09-08:
Doh!  Never mind!  I downloaded the webpage by accident! :/

IdoMagal, 2020-09-08:
Great! :)

Alatarvinyamar, 2020-09-08:
Looks perfect now, thanks!