dppdppd/The-Boardgame-Insert-Toolkit #16: Text label not rendering on lid in last version

yrvn, 2020-09-18:
Hi, I updated the library to the latest version (2.39) and now the text label in the lid doesn't work anymore with the same code. Checked the docs and didn't find anything new about it, is this a bug or something changed?

With v2.17:
[image]

With v.2.39:
[image]

By the way, with the new version the preview looks WAY better.
Thanks!

Code:

```data =
[
  ["Money",
    [
      [ BOX_SIZE_XYZ, [50, 50, 25] ],
      [ BOX_LID,
        [
          [ LID_PATTERN_RADIUS,         5],        
          [ LID_PATTERN_N1,               7 ],
          [ LID_PATTERN_N2,               7  ],
          [ LID_PATTERN_ANGLE,            45 ],
          [ LID_PATTERN_ROW_OFFSET,       10 ],
          [ LID_PATTERN_COL_OFFSET,       140 ],
          [ LID_PATTERN_THICKNESS,        0.6 ],
          [ LID_NOTCHES_B,        t],
          [ LID_FIT_UNDER_B,      t],
          [ LID_CUTOUT_SIDES_4B, [f,f,f,f]],
          [ LID_SOLID_B, f],
        ]
      ],
      [LABEL,
        [
          [ LBL_TEXT,     "Money"],
          [ LBL_SIZE,     AUTO ],
          [ ROTATION,     0 ],
        ]
      ],
      [BOX_COMPONENT,
        [
          [CMP_COMPARTMENT_SIZE_XYZ,  [ 47, 47, 22]],
          [CMP_NUM_COMPARTMENTS_XY,   [1,1]],
          [CMP_SHAPE, FILLET],
          [CMP_CUTOUT_SIDES_4B, [ f,f,f,f] ],
        ]
      ],                            
    ]
  ]
];

IdoMagal, 2020-09-19:
Hi Yuri,

I added support for labeling the box itself, which means that lid labels now below inside the BOX_LID array. Also, you can create any number of labels on the lid and on the box. 

Here is your updated script:

```
[
  ["Money",
    [
      [ BOX_SIZE_XYZ, [50, 50, 25] ],
      [ BOX_LID,
        [
          [ LID_PATTERN_RADIUS,         5],        
          [ LID_PATTERN_N1,               7 ],
          [ LID_PATTERN_N2,               7  ],
          [ LID_PATTERN_ANGLE,            45 ],
          [ LID_PATTERN_ROW_OFFSET,       10 ],
          [ LID_PATTERN_COL_OFFSET,       140 ],
          [ LID_PATTERN_THICKNESS,        0.6 ],
          [ LID_FIT_UNDER_B,      t],
          [ LID_SOLID_B, f],

          [LABEL,
            [
              [ LBL_TEXT,     "Money"],
              [ LBL_SIZE,     AUTO ],
              [ ROTATION,     0 ],
            ]
          ],          
        ]
      ],

      [BOX_COMPONENT,
        [
          [CMP_COMPARTMENT_SIZE_XYZ,  [ 47, 47, 22]],
          [CMP_NUM_COMPARTMENTS_XY,   [1,1]],
          [CMP_SHAPE, FILLET],
          [CMP_CUTOUT_SIDES_4B, [ f,f,f,f] ],
        ]
      ],                            
    ]
  ]
];
```

Good luck on your project!

closed 2020-09-19 by IdoMagal

yrvn, 2020-09-20:
Oh nice! Thank you!

yrvn, 2020-10-14:
Hi @IdoMagal noticed something with this, using that code the label isn't centered in the lid. I guess that's a bug. Tried using [ POSITION_XY,     [0,0]], and same result, I guess that is what it uses if the property isn't set. 

[image]

IdoMagal, 2020-10-15:
Unfortunately, what you have there is what Openscad thinks is centered text. Probably due to the specification in the font itself.
And since Openscad doesn't provide any way to measure fonts or texts, I don't have the ability to center it based on the outer dimensions.

So you'll need to use the position field to fine tune the placement for your chosen font.

To confirm that it isn't a bug of mine, I replaced your text with a simple square. This is what I get:

[image]

yrvn, 2020-10-15:
Thanks for the answer!