dppdppd/The-Boardgame-Insert-Toolkit #18: Cant get this version of the toolkit to work

ebp921, 2020-10-19:
I am able to get the old one on thingiverse to function but the preview dosent seem to change and is extremely fuzzy so I have to render every time I wanna see a change. Someone mentioned something about z ‘axis’? I build foam inserts for myself and friends and I’d really love to be able to speed up the process and this looks like the place to be if I can get it to work.

IdoMagal, 2020-11-27:
What happens when you follow the directions here: https://github.com/IdoMagal/The-Boardgame-Insert-Toolkit#how

closed 2020-11-27 by IdoMagal

ebp921, 2020-11-27:
I got the new version running with help of using someone else’s insert scripts. If I try to start from scratch or use the examples on the page I get a syntax error on line 2 every time. Even if I use code that’s working and remove everything to barebones open and close