dppdppd/The-Boardgame-Insert-Toolkit #19: "Fillet" get's calculated from box bottom, not component bottom

leberschnitzel, 2020-10-29:
Fillets always get calculated from box bottoms and not from component bottoms, that makes them useless if the component part is much higher.
Example Flat Bottom:
[image]

Example Higher Bottom:
[image]

Changing the radius too much causes distortion.
What I would expect is that the bottom of the Fillet always is the bottom of the component.

IdoMagal, 2020-11-27:
fixed in 2.41

closed 2020-11-27 by IdoMagal

leberschnitzel, 2020-11-27:
Perfect, just tested it and works like a charm!