dppdppd/The-Boardgame-Insert-Toolkit #23: Not Really an Issue, no general forum section to ask questions. But Could be Added feature?

ebp921, 2021-01-15:
So im messing with the pedestal options, id really like to be able to essentially have the base of the box flat except for one side that will dip into a fillet style dip. So when you push down on the one side the cards or tokens would pop up from the other side. I realize this is what the pedestal key is for but i cant seem to get it to do what i want and maybe its just from lack of knowledge.

IllyStorm, 2021-05-07:
I'd have found this helpful, I've been learning to use the tool essentially by playing with it, exporting models and figuring out the defaults not specified in the documentation

dppdppd, 2026-02-15:
General questions are welcome in GitHub Discussions or Issues. For the feature mentioned (custom compartment shapes), see CMP_SHAPE options and OpenSCAD's import() for custom SVG/DXF profiles.