dppdppd/The-Boardgame-Insert-Toolkit #38: Box finger cutouts height not working properly

gjkf, 2021-08-16:
When making a box with a cutout like this one
```
[   "cutout box",
        [
            [ BOX_SIZE_XYZ, [40, 57, 36] ],
            [ BOX_COMPONENT,
                [
                    [ CMP_COMPARTMENT_SIZE_XYZ, [ 34, 5, 33] ],
                    [ CMP_CUTOUT_SIDES_4B, [t,f,f,f] ],
                    [ CMP_CUTOUT_HEIGHT_PCT, 77 ], // this line is important
                    [ CMP_CUTOUT_WIDTH_PCT, 30 ],
                    [ CMP_CUTOUT_TYPE, EXTERIOR ],
                ]
            ],
            [ BOX_LID,
                [
                    [ LID_SOLID_B, t],
                ]
            ]
        ]
    ],
```
if instead of 77% the height was anything below 76.388888... the cutout would correctly make a hole on the front and right walls. The reason seems to be that the model used to do the cutout is not centred properly. [Here](https://i.imgur.com/jsqLuvN.png)'s a screenshot of the above model (the 5 size of the compartment is only to show the cutout).

malnoxon, 2023-09-24:
I think the reason is related to the __round_bottom() function `function __round_bottom() = __finger_cutouts_bottom() > m_box_size[ k_z ] - __size( k_z );`

I'm fighting the same problem and find if I just change that to `function __round_bottom() = 0;` the problem is resolved with no adverse effects for my specific use case.

Your case appears to be the opposite of mine, setting `function __round_bottom() = 1;` seems to fix it for the config you posted.

dppdppd, 2026-02-15:
Believed fixed by the asymmetric cutout fix in #65 (commit 17299b5). Tested in v4 with 76%, 77%, and 90% cutout heights — all render correctly with proper rounded-bottom transitions. If the issue persists with current v4, please reopen with a reproducible example.