dppdppd/The-Boardgame-Insert-Toolkit #39: All kinds of visual preview errors

timminkov, 2021-08-24:
Using the latest version of the kit plus the example I get all kinds of weird rendering issues. I've never used OpenSCAD so maybe this is normal but it feels incredibly difficult to use this way. For example, if I render this simple box you can see that there are wall cutouts as intended: 
[image]

However, if I immediately save/preview the same file without any changes, this is what I see:
[image]

And if I rotate the camera around, it feels like there are "artifacts" (look at top left of the lid, for example):
[image]

What am I missing? Is it like this for everyone? Hitting f6 to render each time I need to see my changes will be incredibly slow.

IdoMagal, 2021-08-24:
Previewing subtractive geometry is not great and does result in lots of rendering artifacts, although I do not see the same issues. Have you messed around with the Openscad rendering settings?

timminkov, 2021-08-24:
> Previewing subtractive geometry is not great and does result in lots of rendering artifacts, although I do not see the same issues. Have you messed around with the Openscad rendering settings?

I have. I've also messed around with different Openscad versions to see if there was some kind of mismatch but I see no difference in 2019/2021 or the latest dev build.

Here's the code, for reference:
```
include <boardgame_insert_toolkit_lib.2.scad>;

// determines whether lids are output.
g_b_print_lid = true;

// determines whether boxes are output.
g_b_print_box = true;

// Focus on one box
g_isolated_print_box = "";

// Used to visualize how all of the boxes fit together.
g_b_visualization = false;

// this is the outer wall thickness.
//Default = 1.5mm
g_wall_thickness = 1.5;

// The tolerance value is extra space put between planes of the lid and box that fit together.
// Increase the tolerance to loosen the fit and decrease it to tighten it.
//
// Note that the tolerance is applied exclusively to the lid.
// So if the lid is too tight or too loose, change this value ( up for looser fit, down for tighter fit ) and
// you only need to reprint the lid.
//
// The exception is the stackable box, where the bottom of the box is the lid of the box below,
// in which case the tolerance also affects that box bottom.
//
g_tolerance = 0.15;

// This adjusts the position of the lid detents downward.
// The larger the value, the bigger the gap between the lid and the box.
g_tolerance_detents_pos = 0.1;

data =
[
    [   "Doors",
        [
            [ BOX_SIZE_XYZ,                                     [103, 62, 32.0] ],
            [ BOX_STACKABLE_B, f ],
            [ ENABLED_B, t ],
            [ BOX_LID,
                [
                    [ LID_SOLID_B, f],
                    [ LID_FIT_UNDER_B, f],
                    [ LABEL,
                        [
                            [ LBL_TEXT,     "Doors"],
                            [ LBL_SIZE,     AUTO ],
                        ]
                    ],
                ]
            ],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 31, 27, 30.0] ],
                    [CMP_NUM_COMPARTMENTS_XY,               [3,2] ],
                    [ CMP_CUTOUT_SIDES_4B, [ t, f, f, f ] ],
                ]
            ],
        ]
    ],
];

MakeAll();
```

timminkov, 2021-08-24:
Also, FWIW, none of the other Openscad generic examples seem to have this issue.

timminkov, 2021-08-24:
In addition, it seems like the preview for anything generated using this toolkit slows down the FPS in the preview considerably. I've tried loading a few other people's files and had no issues - Smooth 144 fps.

IdoMagal, 2021-08-24:
So shortcomings of OpenSCAD aside, the reason this is manifesting in your box and not in the examples is because the finger cutouts are butting up against the wall edge, which isn't a situation that was designed for.

Practically, you should only have cutouts on one side and not the other if it's going to be only 1 compartment deep. Otherwise you get a compartment in the second row that you can't actually get your finger in.

My approach would be to either mirror the finger cutouts so they're on both sides, like so:

 ```
   [   "Doors",
        [
            [ BOX_SIZE_XYZ,                                     [103, 62, 32.0] ],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 31, 29, 30.0] ],
                    [CMP_NUM_COMPARTMENTS_XY,               [3,2] ],
                    [ CMP_CUTOUT_SIDES_4B, [ t, t, f, f ] ],
                ]
            ],
        ]
    ],
```
    
    Or if you need cutouts on the front row, make it a 3x1 and create a new component for the back row that has no cutouts.
    
    Hope that helps.

timminkov, 2021-08-24:
Unfortunately changing the mirror cutouts doesn't help. The wall still shows up as solid:
[image]

IdoMagal, 2021-08-25:
What does it look like when you use the compartment dimensions I posted?

timminkov, 2021-08-25:
Ah, that seemed to have at least partially fixed it! Sorry, I didn't catch that you also updated the compartment size in your post. Here's what it looks like now:
[image]

To get around the compartments on all sides thing, I've just set [ CMP_CUTOUT_TYPE, EXTERIOR ] which gets me what I need. Thanks!

closed 2021-08-25 by timminkov

IdoMagal, 2021-08-25:
Excellent!