dppdppd/The-Boardgame-Insert-Toolkit #40: Multiple compartments within a single box not working

timminkov, 2021-08-25:
Hey! Sorry for all the reports - Not sure if I'm doing something wrong here but it seems like it's a very similar to a (fixed) issue here: https://github.com/IdoMagal/The-Boardgame-Insert-Toolkit/issues/2

Using the following code, I'm unable to get a second compartment to appear within a box. Instead, all I see is a rectangle appearing where the compartment _should_ be, almost as if it's adding instead of subtracting:

```
include <boardgame_insert_toolkit_lib.2.scad>;

// determines whether lids are output.
g_b_print_lid = true;

// determines whether boxes are output.
g_b_print_box = true;

// Focus on one box
g_isolated_print_box = "";

// Used to visualize how all of the boxes fit together.
g_b_visualization = false;

// this is the outer wall thickness.
//Default = 1.5mm
g_wall_thickness = 1.5;

// The tolerance value is extra space put between planes of the lid and box that fit together.
// Increase the tolerance to loosen the fit and decrease it to tighten it.
//
// Note that the tolerance is applied exclusively to the lid.
// So if the lid is too tight or too loose, change this value ( up for looser fit, down for tighter fit ) and
// you only need to reprint the lid.
//
// The exception is the stackable box, where the bottom of the box is the lid of the box below,
// in which case the tolerance also affects that box bottom.
//
g_tolerance = 0.15;

// This adjusts the position of the lid detents downward.
// The larger the value, the bigger the gap between the lid and the box.
g_tolerance_detents_pos = 0.1;

data =
[
    [   "Setup",
        [
            [ BOX_SIZE_XYZ,                                     [64, 44, 50.0] ],
            [ BOX_STACKABLE_B, f ],
            [ ENABLED_B, t ],
            [ BOX_LID,
                [
                    [ LID_SOLID_B, f],
                    [ LID_FIT_UNDER_B, f],
                    [ LABEL,
                        [
                            [ LBL_TEXT,     "Setup"],
                            [ LBL_SIZE,     AUTO ],
                        ]
                    ],
                ]
            ],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 9, 38, 48.0] ],
                    [CMP_NUM_COMPARTMENTS_XY,               [3,1] ],
                    [ CMP_PADDING_HEIGHT_ADJUST_XY, [ -3, -3 ] ],
                    [ POSITION_XY, [1.5,CENTER] ]
                ]
            ],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 9, 27, 48.0] ],
                    [CMP_NUM_COMPARTMENTS_XY,               [1,1] ],
                    [ POSITION_XY, [31.5,CENTER] ]
                ]
            ],
        ]
    ],
];

MakeAll();
```

IdoMagal, 2021-08-25:
It looks correct to me?

[image]

timminkov, 2021-08-25:
AH. I had checked every single feature in the settings trying to fix the issue from earlier. Once I reverted those settings the compartment showed up. Sorry for wasting your time :(

closed 2021-08-25 by timminkov

IdoMagal, 2021-08-25:
No worries!