dppdppd/The-Boardgame-Insert-Toolkit #41: Feature request: Specify/limit finger cut-out depth

MrTeamaker, 2021-10-07:
Hi,

As a user of the library, I would like to be able to specify (or limit) the perpendicular size of the finger cut-out, so that I can prevent it from interacting with other box components in the same box.

Currently there's a parameter to adjust theheight and width of the cutout, however there's no way to adjust how much of the box is cut-out perpendiculary (the room into which the fingers need to fit). This is causing me issues as I have some adjacent box elements (of different sizes) and their respective cut-outs are interferring with each other.

I tried to find the right place in the library file  where to at least adjust this parameter, and tracked it down to perp_size, however that in itself was not sufficient as it's having a different effect in each direction.

I'd be happy to help but would need to understand a bit more about the thinking behind how the cut-outs (and their margins) are used/calculated.

IdoMagal, 2021-10-11:
Can you post an example of the interference?

pkensler, 2022-08-15:
I experience this issue frequently. See photo - only the one compartment on the left has a cutout (green) but the finger cutout extends to the right (the "notch" you see circled in red
[image]
)

dppdppd, 2026-02-15:
Fixed in v4 (commit 48dd723). New parameter CMP_CUTOUT_DEPTH_MAX caps the absolute intrusion depth of side cutouts in mm. Set e.g. [CMP_CUTOUT_DEPTH_MAX, 8] to limit cutouts to 8mm depth, preventing interference between adjacent compartments. Default 0 = no cap (existing behavior).

MrTeamaker, 2026-02-20:
Thank you very much :)