dppdppd/The-Boardgame-Insert-Toolkit #45: Box is stackable by default and the BOX_STACKABLE_B is not working

fzerorubigd, 2022-07-23:
This is my simple box : 
```scad
z_border = 2;
z_border_twice = z_border * 2;

player_width = 90;
player_height = 75;
player_depth = 28;

data =
[
    [   "Player Box",
        [
            [ BOX_SIZE_XYZ, [player_width, player_height, player_depth] ],
            [ BOX_STACKABLE_B,false],
            [ BOX_LID,
                [
                    [ LID_SOLID_B, t],
                ]
            ],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ player_width - z_border_twice, player_height - z_border_twice, player_depth - z_border ] ],
                ]
            ], 
        ]
    ],
];
```

I am using `OpenSCAD version 2021.01`, with or without ` [ BOX_STACKABLE_B,false],` the box has the edge on the button. I am using the latest git version of this toolkit. 

Is there anything wrong with my data? or is it a bug, I am not sure to be honest.

fzerorubigd, 2022-07-23:
I used one of the sample in the readme, it is different than the picture in readme. 
[image]

elgringoYan, 2022-07-23:
Hi,

BOX_STACKABLE_B creates a rounded button on the bottom of the box, but also
creates an inset on top of the box, so boxes can be stacked on top of
another.
This image was produced with your code, with BOX_STACKABLE_B = true
[image: image.png]

This one with BOX_STACKABLE_B = false

[image: image.png]

You are not having the same results ?

fzerorubigd, 2022-07-23:
Hi, 

The image is not attached (I guess since you replayed using email) 
Also I opened the example.2.scad in this repository, no change at all (except adding `g_isolated_print_box = "simple box"; 
` to focus on simple box only) , and this is the result :
[image]

elgringoYan, 2022-07-23:
Stackable = true 
[image]

Stackable = false
[image]

fzerorubigd, 2022-07-24:
I am not sure if this is the correct behavior tho. The shot from the readme is different.

light-mooner, 2023-12-16:
i agree with @fzerorubigd , what's the point of have 45 degrees overhang if it's not supposed to be stackable ?

According to the README :

`BOX_STACKABLE_B

value is expected to be a bool and determines whether the base of the box is cut to fit on top of an identically sized box. Note that this requires a printer that can print a 45 degree overhang without supports.`

but the for me, the function works, it's just that it overcomplicate the print by having a 45degrees overhang either case.
When BOX_STACKABLE_B is set to false, the bottom should just be flat and not contains any of those anchor points. or the overhang above it, as the only item of the repo that could fit the bottom is actually a lid. maybe it's meant to put the lid under the box when opened, but than, why isn't there any grooves to be able to remove it easily (as it's the case in the top part.)

dppdppd, 2026-02-15:
Not a bug. The bottom edge visible on non-stackable boxes is the lid-fit-under feature (LID_FIT_UNDER_B defaults to true). Set [LID_FIT_UNDER_B, f] in your BOX_LID to remove it. The stacking feature (BOX_STACKABLE_B) adds corner notches on top, which is a separate mechanism.

closed 2026-02-15 by dppdppd