dppdppd/The-Boardgame-Insert-Toolkit #46: ENABLED_B has no effect on LABEL

fzerorubigd, 2022-08-14:
In this example : 
```scad
data =
[
    [   "simple box",
        [
            [ BOX_SIZE_XYZ,                                     [45, 45, 15.0] ],
            [ BOX_LID,
                [
                    [ LID_SOLID_B, t],
                    [ LABEL,
                        [
                            [ LBL_PLACEMENT, CENTER],
                            [ LBL_TEXT,     "Example"],
                            [ LBL_SIZE,     7],
                            [ ENABLED_B,   f ],
                        ]
                    ],   
                ]
            ],
            [ BOX_COMPONENT,
                [
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 42, 42, 13.0] ],
                ]
            ],                            
        ]
    ],
];

```

The label for the lid should be disabled, but the result has the label. 
[image]

DanForever, 2024-05-05:
I have this problem as well

dppdppd, 2026-02-15:
Fixed in v4 (commit 0359b9f). ENABLED_B inside LABEL entries is now honored at all label rendering sites (lid, box, component labels).

closed 2026-02-15 by dppdppd