dppdppd/The-Boardgame-Insert-Toolkit #57: Text not working in v2.45

Bumblez, 2023-03-21:
Attached I have a simple test file that produced a lid with the label "TEST" on it.

This works fine with BIT lib version 2.11.
But with the latest BIT lib version 2.45, the text no longer appears on the lid!

[BIT_lid_test.scad.txt](https://github.com/dppdppd/The-Boardgame-Insert-Toolkit/files/11034321/BIT_lid_test.scad.txt)

rd00169438, 2023-05-16:
I had a similar problem where I couldn't get text on a lid, so I tried out the Dune example shown in the readme. This works with it's supplied toolkit version (2.11), but not in the current version.

I have tried different versions from the history and found it stops working as of version 2.34.

rd00169438, 2023-05-16:
The issue seems to be a change in 2.34 to make the lid labels a sub-object of BOX_LID.

The example in the readme hasn't been updated to match and needs to nest the labels under the lid like this:

        [ BOX_LID,
            [
                [ LID_FIT_UNDER_B,      f],
                [ LID_CUTOUT_SIDES_4B, [f,f,t,t]],
                [ LID_SOLID_B, f],
                [ LID_HEIGHT, 15 ],
                [ LABEL,
                    [
                        [ LBL_TEXT,     "Skull     and"],
                        [ LBL_SIZE,     AUTO ],
                        [ ROTATION,     45 ],
                        [ POSITION_XY, [ 2,-2]],
                    ]
                ],
                [ LABEL,
                    [
                        [ LBL_TEXT,     "Crossbones"],
                        [ LBL_SIZE,     AUTO ],
                        [ ROTATION,     315 ],
                        [ POSITION_XY, [ -4,-0]],
                    ]
                ],        
            ]
       ],

Another way to get BIT files with the older structure to work with more recent versions of the toolkit it to change the line:

    m_lid = __value( box, BOX_LID, default = [] );

to:

    m_lid = box;

In boardgame_insert_toolkit_lib.2.scad, which reverts the structure back to label under box rather than lid.

Genosystem, 2023-08-05:
> The issue seems to be a change in 2.34 to make the lid labels a sub-object of BOX_LID.
>
> The example in the readme hasn't been updated to match and needs to nest the labels under the lid like this: [...]

Thanks @rd00169438 , nesting LABEL worked for me

dppdppd, 2026-02-15:
v2 is no longer maintained. Please use v4 which has working text/label support.

closed 2026-02-15 by dppdppd