dppdppd/The-Boardgame-Insert-Toolkit #58: Is this still active?

rtrout, 2023-05-15:
I'm new to OpenSCAD and trying out the BIT. I'm having trouble getting started with BIT, and am unsure if I'm doing something wrong, or if BIT is broken.

Trying to follow the 'HOW' instructions, and starting with the code below.

I get a bunch of warnings from OpenSCAD when trying to refresh, like:

> WARNING: undefined operation (undefined - number) in file boardgame_insert_toolkit_lib.2.scad, line 750 
> WARNING: undefined operation (undefined - number) in file boardgame_insert_toolkit_lib.2.scad, line 750 
> WARNING: undefined operation (undefined + number) in file boardgame_insert_toolkit_lib.2.scad, line 1874 

It would help if there was some kind of basic example I could start with and confirm working, then add to. Based on the project page, I can't work out the minimum code required to get started.

Hints and tips?

```
include <boardgame_insert_toolkit_lib.2.scad>;

data = 
[   "example 0: minimal",                            // our box. name is just for code organization.
    [
        [ BOX_SIZE_XYZ, [46.5, 46.5, 15.0] ],        // one kv pair specifying the x, y, and z of our box exterior.
        [ BOX_COMPONENT,                             // our first component.
            [
                [ CMP_NUM_COMPARTMENTS_XY, [4, 4] ],               // it's a grid of 4 x 4
                [ CMP_COMPARTMENT_SIZE_XYZ, [ 10, 10, 13.0] ],   // each compartment is 10mm x 10mm x 13mm
            ]
        ]
    ]
];

MakeAll();
```

HobbitHardcase, 2023-05-26:
You should start with the examples.2.scad file, as listed in the instructions. If you just want a minimal box, then example_mmu.scad works as well. The code you have listed above doesn't.

rtrout, 2023-05-27:
> You should start with the examples.2.scad file, as listed in the instructions. If you just want a minimal box, then example_mmu.scad works as well. The code you have listed above doesn't.

Thanks. I ended up getting my head around Fusion 360. If I want to return to this, at least I'll know now where to start. Thanks.

dppdppd, 2026-02-15:
Yes, actively maintained. v4 released with significant improvements. See README.