dppdppd/The-Boardgame-Insert-Toolkit #60: Labels on solid lids not respecting depth?

GregRMitchell, 2023-08-23:
I'm attempting to build a box for Glory to Rome, which came in an awful clamshell package.  I put a label on the solid lid with the game name, and it doesn't seem to be paying any attention to the depth settings, instead being a complete void through the whole lid.  

Here's my lid code:
```
            [ BOX_LID,
                [
                    [ LID_INSET_B, t],
                    [ LID_SOLID_B, t],
                    [ LID_HEIGHT, 4 ],
                    [ LID_SOLID_LABELS_DEPTH, 1],
                    [ LABEL, 
                        [
                            [LBL_TEXT, 
                                "Glory to Rome"
                            ],
                            [LBL_DEPTH, 1],
                            [ LBL_SIZE, AUTO],
                            [ LBL_FONT, "Times New Roman:style=bold"]
                        ]

                    ],
                ],
                
            ],  
```

Am I doing something wrong, or is it broken?

dvirbuc, 2024-05-16:
I have faced the same problem, to fix it I changed the "boardgame_insert_toolkit_lib.2.scad" file a little.
from (line 1745):
`MakeAllLidLabels( thickness = thickness );`

to:
```
if (m_has_solid_lid && m_lid_label_depth > 0 )
	MakeAllLidLabels( thickness = thickness - m_lid_label_depth );
else
	MakeAllLidLabels( thickness = thickness );
```
I don't know if it breaks other stuff, but for a simple box, it's working.

file for reference:
[solid_lid_with_label.scad.txt](https://github.com/dppdppd/The-Boardgame-Insert-Toolkit/files/15335276/solid_lid_with_label.scad.txt)
[boardgame_insert_toolkit_lib.2.scad.txt](https://github.com/dppdppd/The-Boardgame-Insert-Toolkit/files/15335277/boardgame_insert_toolkit_lib.2.scad.txt)

anodynesoftware, 2025-11-27:
The following is the equivalent fix for v3 of the library. Three points to note:
1. The fix needs applying in two places in the .scad file, once for regular boxes and once for hex boxes. I have only tested the fix on regular boxes, but believe that it applies equally to hex boxes.
2. The key LBL_DEPTH is ignored for labels in solid lids.
3. The key to use for solid lids is LID_SOLID_LABELS_DEPTH, and this must be set to the depth required for the label. For example, a depth of 0.1 is barely visible, whereas a depth of 3 on a 2mm lid would be all the way through. Note that in the fix above, this key instead defines the thickness of lid remaining at the bottom of the incised text, which is counter-intuitive for me.
Around lines 1785 and 3889 in boardgame_insert_toolkit_lib.3.scad, change:
`        MakeAllLidLabels( thickness = thickness );`
to:
```
    if (m_has_solid_lid)
        MakeAllLidLabels( thickness = m_lid_label_depth );
    else
        MakeAllLidLabels( thickness = thickness );
```

closed 2026-02-15 by dppdppd