dppdppd/The-Boardgame-Insert-Toolkit #63: Text in Label not connected with lid

eingemaischt, 2023-09-10:
Hi, with this code, the label is not connected to the lid. After the print, they just fell of.

That did I do wrong - or is this a bug?

```
include <../The-Boardgame-Insert-Toolkit/boardgame_insert_toolkit_lib.2.scad>;

g_b_print_lid = true;
g_b_print_box = true; 
g_isolated_print_box = ""; 
g_b_visualization = false;          
//changed from 1.5 to 1
g_wall_thickness = 1;
g_tolerance = 0.15;
g_tolerance_detents_pos = 0.1;

// 60x110mm zusammen
// 75mm hoch

data = 
[
    [   "Test",
        [
            [ BOX_SIZE_XYZ,                                     [128, 77, 15] ],
            [ BOX_STACKABLE_B, t],

            [ BOX_LID,
                [                            [ LABEL,
                [
                    [ LBL_TEXT,     "Bzzz..."],
                    [ LBL_SIZE,     AUTO ],
                    [ LBL_PLACEMENT,     FRONT ],
             
                ],
                    [ LID_SOLID_B, f],
                    [ LID_INSET_B, t],
                ],

            ],
            ],

            [ BOX_COMPONENT,
                [
                    [ CMP_NUM_COMPARTMENTS_XY, [2, 1] ],
                    [CMP_COMPARTMENT_SIZE_XYZ,  [ 62, 74, 13,5] ],
                ]
            ],                            
        ]
    ],

];

MakeAll();
```

eingemaischt, 2024-01-03:
It seems that this Problem only occurs with "[ BOX_STACKABLE_B, t],".

Does anyone knows how to fix that? I don't speak "OpenSCAD" enough ;-)

snarky1, 2024-05-16:
If you go to line 1791 you will find:
`                            if ( g_print_mmu_layer != "mmu_box_layer" )
                                MakeAllLidLabels(); `

Change it to 
`                            if ( g_print_mmu_layer != "mmu_box_layer" )
                                MakeAllLidLabels(thickness = m_lid_thickness +0.5 ); `

And you will have a 0.5mm overlap between the label and the BG.

closed 2026-02-15 by dppdppd