dppdppd/The-Boardgame-Insert-Toolkit #80: Could you tell me how you achieve custom storage for irregular-shaped tokens?

wahaha1993, 2025-01-04:
like the catan boat that shown in the readme page

jdnowak, 2025-02-09:
you make sum of different cuts, for example if you want to have let's say one box with 3 different shapes inside you just add additionall sections for with BOX_COMPONENT that overlay each other if you with, most possibilities you have in bit_functions_lib, and most this functions have llx and lly to set where he has to start cutting from box. Please check this files and go down and find from which of them you can make your box.
To summary BOX_SIZE_XYZ creates box and each BOX_COMPONENT cuts desired item from it.

okainov, 2025-09-22:
@wahaha1993 were you able to find a solution?

@jdnowak sorry, this doesn't help. The question was specifically about irregular shapes. Of course one can, given enough patience, make boat-like shape out of ~50 squares and circles, but that's insane. And I'm quite sure that's NOT how the irregular shape from the readme was achieved.

okainov, 2025-09-24:
@wahaha1993 so far what I came up with it either do Scribble on top of photo overlay in Tinkercad and then extrude the shape as you need (more manual, but might be fast enough) or make a proper photo, then convert it to SVG in Inkscape: Path -> Trace Bitmap, clean/simplify as needed (Path -> Simplify), and then you can import SVG into Tinkercad to extrude.

The problem is to get the right sizes, as for custom shape it's not easy to measure correctly. So I usually do a bunch of test prints, extruding the shape of different sizes from some random circle/rectangle, print it to see how it fits, and then you'll know the right size. SO you should be able to export the right sized model into STL from Tinkercad, as well as actually extrude it from more meaningful stuff (e.g. organizer boxes :))

Here are my couple of examples:

[image]
[image]

closed 2026-02-15 by dppdppd